enjoyneering/ESP8266-I2C-Driver #1: Ошибка компиляции для платы Generic ESP8266 Module.

selan61, 2018-01-03:
Выполнил замену.
Copy and relpace "twi.h", "core_esp8266_si2c.c" in folder %USERPROFILE%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266
Copy and replace "Wire.h", "Wire.cpp" in folder %USERPROFILE%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire

Пытаюсь подключить SI7021 к ESP01 на пины 0,2. Беру Ваш пример из библиотеки HTU21D-master SI7021_DEMO при компиляции выдает ошибку.

C:\Users\HOME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\_core_esp8266_si2c.c:157:15: error: conflicting types for 'twi_writeTo'

unsigned char twi_writeTo(unsigned char address, unsigned char * buf, unsigned int len, unsigned char sendStop){

^

In file included from C:\Users\HOME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\_core_esp8266_si2c.c:21:0:

C:\Users\HOME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\twi.h:66:9: note: previous declaration of 'twi_writeTo' was here

uint8_t twi_writeTo(uint8_t address, uint8_t *buffer, uint8_t length, bool sendStop);

^

C:\Users\HOME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\_core_esp8266_si2c.c:181:15: error: conflicting types for 'twi_readFrom'

unsigned char twi_readFrom(unsigned char address, unsigned char* buf, unsigned int len, unsigned char sendStop){

^

In file included from C:\Users\HOME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\_core_esp8266_si2c.c:21:0:

C:\Users\HOME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\twi.h:67:9: note: previous declaration of 'twi_readFrom' was here

uint8_t twi_readFrom(uint8_t address, uint8_t *buffer, uint8_t length, bool sendStop);

^

Используем библиотеку Wire версии 1.0 из папки: C:\Users\HOME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire 
Используем библиотеку HTU21D-master версии 1.1.0 из папки: D:\Arduino\libraries\libraries\HTU21D-master 
exit status 1
Ошибка компиляции для платы Generic ESP8266 Module.

enjoyneering, 2018-02-10:
вы не до конца/не правильно выполнили замену. повторите все еще раз.

dontsovcmc, 2018-04-15:
А есть где-то diff, чтобы посмотреть Ваши изменения? 
У меня Wire плохо работает с Attiny85 1Mhz: ESP-01 спрашивает у него по байтам буфер. Где-то после 20-30 байта возвращается FF FF.  Ваше лекарство мне подойдет? =)

closed 2018-06-11 by enjoyneering

enjoyneering, 2018-08-01:
diff нет и не будет. я слишком ленивый для этого.
теоритический должно помочь. чтоб в этом убедится надо попробовать.

dontsovcmc, 2018-08-01:
я решил эту проблему! в wire для ESP надо было увеличить время ожидания бита. а с avr проблемы нет.

enjoyneering, 2018-08-07:
вы имете в виду setClockStretchLimit()? и до скольки увеличивали? по умолчанию забито 230

dontsovcmc, 2018-08-07:
Да! До 1500.  Нижний предел не помню. Так надежно работает.
https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/master_i2c.cpp#L12

enjoyneering, 2018-08-13:
на сколько я знаю у Attiny85 нету железного I2C. какой **soft** Slave I2C либой пользовались для Attiny85?

dontsovcmc, 2018-08-13:
Вот здесь и master, и slave. https://github.com/dontsovcmc/USIWire

enjoyneering, 2018-08-15:
не пойму зачем если по arduino есть фреймворк для ATTinyCore там есть все и master I2C и slave I2C.
находится тут - https://github.com/SpenceKonde/ATTinyCore

enjoyneering, 2018-08-22:
обманул вас. оказывается Arduino IDE линкует библиотечные файлы перед скечем. те вставка ПЕРЕД #include <Wire.h> добавьте #define TWI_I2C_BUFFER_LENGTH 40 в теле *.ino файла ничего не даст. надо ручками править define в библиотеке.

dontsovcmc, 2018-08-22:
Я поясню, почему другую библиотеку взял: я сначала повторил чужой проект переостанции, а там было переключение i2c Master, потом Slave и возможно это не поддерживает стандартный wire. А потом уже не выкинул. Надо проверить работу wire, спасибо!