enjoyneering/ESP8266-I2C-Driver #2: open drain on I2C GPIO pins

baradoid, 2018-10-20:
your library works ok, but why you don`t use open drain on GPIO ?

> #define SDA_LOW()  (GPES = (1 << twi_sda))        //enable  SDA (becomes OUTPUT       & since GPO is 0 for the pin, it will pull the line low)
> #define SDA_HIGH() (GPEC = (1 << twi_sda))        //disable SDA (becomes INPUT_PULLUP & since it has 30kOhm..100kOhm pullup it will go high)
> #define SDA_READ() ((GPI & (1 << twi_sda)) != 0)
> #define SCL_LOW()  (GPES = (1 << twi_scl))
> #define SCL_HIGH() (GPEC = (1 << twi_scl))
> #define SCL_READ() ((GPI & (1 << twi_scl)) != 0)

you can use it as described in "ESP8266 Technical Reference". All useful defines in esp8266/esp8266_peri.h

> 
> //GPIO (0-15) PIN Control Registers
> #define GPC(p) ESP8266_REG(0x328 + ((p & 0xF) * 4))
> #define GPC0   ESP8266_REG(0x328) //GPIO_PIN0
> #define GPC1   ESP8266_REG(0x32C) //GPIO_PIN1
> #define GPC2   ESP8266_REG(0x330) //GPIO_PIN2
> #define GPC3   ESP8266_REG(0x334) //GPIO_PIN3
> 
> #define GPC11  ESP8266_REG(0x354) //GPIO_PIN11
> #define GPC12  ESP8266_REG(0x358) //GPIO_PIN12
> #define GPC13  ESP8266_REG(0x35C) //GPIO_PIN13
> #define GPC14  ESP8266_REG(0x360) //GPIO_PIN14
> #define GPC15  ESP8266_REG(0x364) //GPIO_PIN15
> 

> 
> //GPIO (0-15) PIN Control Bits
> #define GPCWE  10 //WAKEUP_ENABLE (can be 1 only when INT_TYPE is high or low)
> #define GPCI   7  //INT_TYPE (3bits) 0:disable,1:rising,2:falling,3:change,4:low,5:high
> #define GPCD   2  //DRIVER 0:normal,1:open drain
> #define GPCS   0  //SOURCE 0:GPIO_DATA,1:SigmaDelta
>

enjoyneering, 2018-10-24:
Hi,

Someone has already asked this question to Ivan Grokhotkov, but I can't find his answer. All I remember, he used  INPUT_PULLUP in Arduino ESP8266 core because many people do not know about I2C Bus Pullup Resistor.

enjoyneering, 2018-12-06:
found the comment:

> "Due to lack of documentation regarding pin circuits from Espressif it’s really hard to say whats going on inside the chip. However I did some tests and it looks like ESP8266 pins do have an open-drain output mode. In Arduino you can enable it by calling pinMode(pin, OUTPUT_OPEN_DRAIN); on the pin of your choice. This mode was used in the original version of our software I2C code.
**However we had issues from people forgetting about external pull up resistors, so the code has changed a bit recently.** Instead of changing pin output register when banging the bits, we now change pin mode register. The pin is switched between INPUT_PULLUP and OUTPUT modes, which makes it either a weak pull up or a strong pull down (output register has 0 written into it in advance).

closed 2019-01-15 by enjoyneering