enjoyneering/ESP8266-I2C-Driver #6: Не совместимость с последней версией ядра

dzanis, 2019-08-14:
У меня проблемы с подкючением DS3231 на ESP-01 , в начале не хочет читать время,только позже и то надо вручную это делать через веб интерфейс,что не подходит,ведь надо после включения автоматом считать время.Попробовал ваш фикс I2C ,и сразу выскочило несколько ошибок,исправить недолго,не было difine TWI_BUFFER_LENGTH и методах twi_writeTo twi_readFrom аргументы менял на unsigned char .Но RTC DS3231 вообще не работал,даже вручную.Думаю может ошибка в загруженности ESP-01 в начале инициализации.Как думаете они уже пофиксили в официальном ядре I2C?

enjoyneering, 2019-08-26:
странно. сам проврял на новом ядре 2.5.2 - никаких ошибок и конфликтов. менять надо ВСЕ файлы, старые УДАЛЯТЬ или переименовать **расширение**. если поменять название то не заработает. потому что линковщик работает с тем что внутри файлов *.h и *.cpp, а не с именами.

у ESP-01 не выведенны SDA - GPIO4 и SCL - GPIO5, а те что есть GPIO1 и GPIO3 заняты Serial куда постоянно валятся служебные сообщения и все то что вы оправняете командой Serial.print()

есть способ  переназначить Serial с GPIO1 и GPIO3 на GPIO15 и GPIO13:

`Serial.swap();  //remap serial pins GPIO1/D2/UART0TXD & GPIO3/D7/UART0RXD to GPIO15/UART0RTS & GPIO13/UART0CTS`

после этого скорее всего придется модифицировать библиотеку DS3231. какой вы пользуетесь?

dzanis, 2019-08-26:
Да,а вот версию я не посмотрел что у меня не последняя  лол :) Версия ядра ESP  у меня 2.5.0  Arduino IDE 1.8.9
Да, не выведены GPIO4 и GPIO5,поэтому менял подключения
```
 //вместо rtc.begin(); пишем
Wire.begin(0, 2);
```
Т.е модифицировать библиотеку DS3231 не нужно,просто Wire.begin( sda, scl);
Решил проблему "в начале не хочет читать время" тем что делаю несколько опросов RTC, и тогда всё нормально.

dzanis, 2019-08-26:
Вот сериал мне как раз таки был нужен,для дебага.Сейчас его переделывать,сложно,на плате стоит не у меня.Библиотеку  для DS3231  брал от сюда  https://github.com/adafruit/RTClib
Драйвер не заработал,вообще не читало время, но это понятно,т.к версия ядра другая,как уже писал выше.Но позже думаю обновить ядро и попробывать ваш драйвер.Да надо разбираться,но у меня нету логического анализатора,и как без него разбираться в том чего не видно.Вот только методом проб удалось добиться решения,хоть и костыльное оно. Сейчас уже то устройство запаяно,работает у друга дома,забирать я его не буду.Надо новое собирать,и если настрою с вашим драёвером,то просто обновлю еспешки.