enjoyneering/ESP8266-I2C-Driver #7: The 'clock stretch limit' is artificial, not part of the I2C spec

Tech-TX, 2019-09-05:
I realize it's been part of the code since early on, but it's a _Bad Idea_.  There is NO requirement in any version of the I2C spec I've seen that describes a limit on clock stretch timing.  In fact, the spec clearly says that the bus theoretically goes to DC (0Hz), depending on the requirements of the slave.  SMBus has a minimum speed of 10KHz, but I2C has no lower limit.  I've used I2C parts before that added **MUCH** more stretch than is listed in this driver (230uS).  By introducing an artificial and arbitrary limit on slave devices, you guarantee that a few people will have issues, and they won't know why without a DSO and some understanding of the I2C protocol.

core_esp8266_si2c.cpp line 215,
twi_setClockStretchLimit(230);      //stretch SCL limit, in μsec

Here's the current ver. 6 of the I2C spec:
https://www.nxp.com/docs/en/user-guide/UM10204.pdf

If you're 'fixing' or 'improving' the original code, you should fix this as well.  I did a quick Google and got 88 hits for "clock stretch limit site:arduino.cc", and at least a few of them are "it works on an Arduino but not on an ESP8266".  That's because the Arduino Wire library has no artificial limit.  I can't imagine a slave that's slow enough to trip the WDT on the ESP with an unusually long stretch, but I suppose it's possible with a bit-bang chunk of code and unknown parts.

enjoyneering, 2019-09-06:
Hi, 

Thanks for the answer. I think the Arduino ESP8266 team added a limit to prevent WDT. Theoretically, something could happen to slave and infinite stretch will restart the ESP8266.  What do you think?

Tech-TX, 2019-09-12:
I've seen 300 millisecond clock stretch with (much older) successive approximation A/D converters.  Looks like some of them nowadays get around that by giving you the PREVIOUS conversion result when you query the chip so it doesn't freeze the I2C bus for a long time.  A MAX11612 does clock stretching with sub-10KHz sample rate (up to 100KHz), so likely 100 to 200us stretch.  Looking at other technologies, one TI delta-sigma ADC (ADS1000) runs 128 samples per second, also returning the previous conversion.  All bets are off with the Chinese clone parts, though.

I realize there should be a WDT feed during unusually long clock stretches, but the bus is allowed to run quite literally to DC, although I've never seen anything longer than 300ms myself 10+ years ago.  I saw people at arduino.cc setting the limit to 1500us or less for various sensors, so I'd probably use that as a default stretch limit before you decide the bus is locked and reset the bus.  1500us would at least minimize the number of people having the "but it works on my Arduino" complaint.

Tech-TX, 2019-09-12:
Here's a more extreme case, and a possible "please everyone" solution:
http://www.digole.com/forum.php?topicID=777

Apparently people were setting the limit to 40ms to get their touchscreens to work, and even that much wasn't always enough.  So, change the default stretch limit to 1500us to please most people, and then use something like the following:
`static bool twi_read_bit(void) {
  uint32_t i = 0;
  SCL_LOW();
  SDA_HIGH();
  twi_delay(twi_dcount+2);
  SCL_HIGH();
  if (twi_clockStretchLimit > 0) {
    while (SCL_READ() == 0 && (i++) < twi_clockStretchLimit);  // Clock stretching
    } else {
    while (SCL_READ() == 0) yield();  // infinite clock stretching, NO bus lock check
  }
  bool bit = SDA_READ();
  twi_delay(twi_dcount);
  return bit;
}`
so that if the user sets Wire.setClockStretchLimit(0); then there's _no_ stretch limit and _no_ bus lock check; the routine hangs if the bus is locked.  I've seen bus lock before with two slaves that refused to play nice together, so we had to move one of them onto a different I2C bus.

enjoyneering, 2019-09-14:
Hi,

Added 10KHz, 15KHz, 25KHz, 50KHz:
`Wire.setClock(10000); //10KHz clock`

Added:
`Wire.setClockStretchLimit(0); //can slow down up to 0Hz & block SCL line forever`

Tech-TX, 2019-09-16:
What I'm seeing in this release is a default limit of 750uS,
in twi.h: #define TWI_I2C_SCL_STRCH_LIMIT         250    //maximum SCL stretch time, in μsec
and with an 80MHz clock in core_esp8266_si2c.cpp:
#define TWI_CLOCK_STRETCH_MULTIPLIER 3  // mult = 6 for 160MHz clock
unless I'm reading your code incorrectly... it happens. ;-)

I have one of the AMS CCS811 VOC sensors I haven't played with yet and I know it stretches the clock, causing issues with the current ESP driver.  I'll try out your current release and let you know the results.

enjoyneering, 2019-09-16:
no, default is about 250us for both MCU clocks. the MULTIPLIER is required to match the empty MCU cycle length with real microseconds.

I've added Wire.setClockStretchLimit(**0**). With a value of zero, you will not get limited clock stretching. The driver will wait for the sensor response forever. I personally do not like it because of the watchdog timer. That is why the 250us/4KHz is still there. But I respect the standard and added 0Hz option for advanced users like you.

Tech-TX, 2019-09-17:
Gotcha on the multiplier... I was just guessing... and guessed wrong. ;-)
I have a good test case for your code, but it may have to wait until this weekend when I have time as I _know_ I'll have to fire up the old H-P MSO to see what the bus is doing.  The SparkFun library for the CCS811 apparently doesn't work AT ALL with the ESP chip.  If your library makes that combo work, then your code is pretty solid.  I've never turned these chips on, so I'm going in with multiple unknowns (broken library, unknown status of chips on a no-name clone board out of China), which isn't a smart way to begin.  Someone else said the Adafruit library works with ESP and CCS811, so I'll likely start there to at least insure the chip responds.  There's no mention in the CCS811 datasheet about clock stretch, but I'm guessing that's what it's doing that breaks the ESP wire library.  From the simplified block diagram, that micro in the chip could be doing *anything*.  https://i.imgur.com/GnvFAs9.jpg  The I2C diagrams in the datasheet are purely state levels; they never mention timing anywhere, and then refer you to the I2C spec.  <rolls eyes>

enjoyneering, 2019-09-17:
Good,

Thank you for your help. The Adafruit CCS811 library has a stretch limit of 500us, see below. To overwrite this number, add Wire.setClockStretchLimit(0) after starting ALL i2c sensors in setup().

`	Wire.begin();
	#ifdef ESP8266
	Wire.setClockStretchLimit(500);
	#endif
`

Tech-TX, 2019-09-22:
Wow... yes, I see that.  I have 3 different versions of that datasheet, and didn't notice that, thanks!  It wasn't in any of the earlier versions of the datasheet.

The worst I'm seeing with a couple of different libraries is around 40us.  The board I'm using has 10K pull-up resistors on the I2C lines, so I ran 330 ohm resistors between the '8266 I2C pins and the CCS811, monitoring at the '811 directly.  When the '811 stretches the clock the level is 60mv, and when the '8266 is driving it's 208mv, so I can see the stretch directly on the 'scope.  I'm only seeing between 10us and 21us stretch normally, nothing wider except for a very rare 40us (couldn't capture it).

[image]

The SparkFun library is working for me without changes using the original wire library, so I'll go ahead and install your changes and see if I can break them.  :-)  I'm pretty good at breaking things.

Tech-TX, 2019-09-22:
Never mind, there's something weird here.  Looks like 'LucAce' borrowed the core parts of the Adafruit library, as Adafruit's library had the same compiler warning in the multi-byte read..  I'd commented the 'unused' variable in the LucAce driver early on (I hate compiler warnings).  I compiled the Adafruit example without that and it worked.  Then I went back and uncommented the 'unused' variable line in the LucAce driver and it also worked.  It's line 251 in the Adafruit_CCS811.cpp file.  I thought it was a local unused variable, but apparently it's passed to the wire library, even though I don't see it initialized anywhere.

Your library is working so far; I'll try a few other things I have here.  I'd *really* like to see the default clock stretch moved up to 1500us instead of 250us.  Your comments say 1250, but your twi.h line 49 still has the old 250us limit.  I've moved it up to 1500 in what I'm testing here to see if Arduino stuff will work on the ESP without errors.  I'm pretending I'm a clueless newbie with I2C. :-)

Tech-TX, 2019-09-23:
Next up to bat: the Bosch BME280.  I was going to do an HD44780, but it looks like you have that covered. ;-)
I'd tried 3 or 4 different libraries for the BME280 a couple of months ago on another ESP8266 project, and settled on the SparkFun lib 'cos it didn't have GPL pollution and was easy to figure out.  Crash and burn.  It acts like it's not recognizing the module at all.  I recompiled a simple I2C scanner from farmerkeith and that was sort of working, but not right.  It knew that something was at the correct address, but it got the device ID wrong.  Recompiled with the old library and it worked again.  Sigh.  Another bug hunt for another day.

Tech-TX, 2019-09-25:
I've been 'round and 'round this little example, and can't find where I'm making a newbie error.  It compiles without warnings and works as expected with the original wire library, but your library returns a zero for the ID byte, and not 0x60.  I tried finding an example you'd done that showed a single byte read, but couldn't find one.

`/* --------------------------------------
  Scan and Identify I2C devices by farmerkeith
  21 January 2019

  Recognised devices are listed as their device type */

#include "Wire.h"

void setup() {
  Serial.begin(9600);
  Serial.print("\nI2C Scan_ID");
  Serial.println("\nScans for I2C devices and names any that it knows about");
  Wire.begin();
}

void loop() {
  uint8_t error, address;
  uint8_t readbytes = 1;
  Serial.println("Scanning...");

  address = 118;

  Wire.beginTransmission(address);
  error = Wire.endTransmission();
  if (error == 0) {
    Serial.print("I2C device found at address 0x");
    if (address < 16) Serial.print("0");
    Serial.print(address, HEX);
    Serial.print(" !");
  } else if (error == 4) {
    Serial.print("Unknown error at address 0x");
    if (address < 16) Serial.print("0");
    Serial.println(address, HEX);
  }
  if (error == 0) {
    uint8_t data1 = 0;
    switch (address) {
      case 0x76: case 0x77:
        Serial.println(" BMP280 or BME280");
        // note: address 0x77 may be BMP085,BMA180 and may not be MS5607 or MS5637 CHECK
        Wire.beginTransmission(address);
        // Select register
        Wire.write(0xD0); // 0xD0 hex address of ID
        // Stop I2C Transmission
        Wire.endTransmission();
        // Request 1 bytes of data
        Wire.requestFrom(address, readbytes, true);
        // Read 1 byte of data
        if (Wire.available() == 1)  {
          data1 = Wire.read();  // **this returns zero with your library, 0x60 with original library**
        } // end of if (Wire.available() == 3)
        Serial.print ("Device ID=");
        Serial.print(data1, HEX);
        if (data1 == 0x58) Serial.println(" = BMP280");
        else if (data1 == 0x60) Serial.println(" = BME280");
        else Serial.println(" ID not in list");
        break;
      default:
        Serial.println("device not in list");
        break;
    }
  }
  delay(5000);           // wait 5 seconds for next scan
}`

Here's what it _should_ be doing, and proof that your library is doing the bit banging perfectly:
[image]
[image]
[image]
[image]

That last capture is zoomed in on the register read (the last sequence).  It's utterly perfect.

Tech-TX, 2019-09-25:
Hrm.  I found a 'fix', although it shouldn't be needed as there's plenty of time after the stop condition from the **Wire.endTransmission();**
https://arduino.stackexchange.com/questions/43007/why-is-a-delay1-necessary-before-wire-requestfrom
I added **delayMicroseconds(1);** before the **Wire.requestFrom()** and it works as expected, returning 0x60 as the ID byte.

enjoyneering, 2019-09-25:
Hi, Tech-TX

Thank you for update. Will check the 0x60. I have a couple of suggestions for your sketch.

Looks like the variables "error", "address", "readbyte" are [global](https://www.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/scope/). Put them before setup().

Another common mistake of all Arduino i2c scanners is starting from 0x00. According to UM10204, 0x00 is the "General call address" for resetting all devices that support this feature. Also the addresses 0x01, 0x02, 0x03 are also reserved.

Use the F() macro for print() class, this will save the text in flash memory, otherwise it will remain in RAM and fragment it

`Serial.println(F("BMP280 or BME280"));`

In your sketch, the "address" & "readbytes" variables never change. To save 2 byte of RAM, you can declare it as #define before "#include "Wire.h"". The #define - it is always written in capital letters, it is a marker indicating that it is not a variable.

`#define ADDRESS 118 //0x76`

and use it
`Wire.beginTransmission(ADDRESS);`

Tech-TX, 2019-09-27:
Additional test: I changed the Wire.requestFrom() to Serial.println(Wire.requestFrom()); and that's also returning zero (no bytes read), as far as I can see by reading twi_readFrom() in the core file, even though it did return the read byte (0x60) in the buffer.

enjoyneering, 2019-09-27:
Hi, Tech-TX

good catch !!! found an error and updated the library

`if (length < 1) return 0;  //nothing to read`

1<1 return false, see [arduino reference](https://www.arduino.cc/reference/en/language/structure/comparison-operators/lessthan/)

updated the library

Tech-TX, 2019-09-28:
Oh, the error isn't 0x60, it's any I2C read (single byte or multi-byte).  I get the same errors on all of them.  I had the same error reading the data from an INA219 current monitor just now with different code.  The data was there but the buffer depth read empty, and I had the other error flags as well.  I can do Serial.println from inside wire.cpp to see what's happening but that doesn't work inside the core file.

enjoyneering, 2019-09-29:
Werd. I've just check my new I2C with BMP180 - works good. The BMP180 has Wire.available()==1 & Wire.available()==2. No issues. 

Did you change all files?

Tech-TX, 2019-10-05:
I'm sooooo confused.  I copied over your latest tweaks and recompiled the little I2C scanner, taking out the _Wire.setClockStretchLimit(1500)_ I'd added since I see you've boosted the limit in **twi.h** to **1250**.  It recompiled without errors, but the serial monitor was showing it was reading 7F for the ID byte in the BMP280.  I shut it down and went to bed.  A couple of days later, I turn everything back on again after fixing my Bus Pirate, and without doing anything it's reporting the correct ID byte.  I didn't change anything, other than a power cycle.  No recompile, no reload, just a power cycle.
I compiled a little CCS811 example just now and it works as well (removing the _setClockStretchLimit_ in his code).  I'll continue with the testing this weekend.  I have a more complex BME280 chunk in a different app I've been running for a few months, and I'll try that with your library as well.
Thanks!

Tech-TX, 2019-10-05:
Grrr.  Broken again, after a few recompiles and power cycles (my Win10 laptop freaks out and hangs the USB ports periodically...).  Am I doing it incorrectly by simply over-writing all 6 of the files with your version while the IDE is shut down?  I have the "Aggressively cache compiled core" unchecked so it does a full recompile each time.  Is there something else I need to do to either clear the cache or register the changed files in the core library?

I'm back to not getting the data available (it's zero), although the read is happening.  I'm using a download of the files from Oct. 3, after your latest changes.

enjoyneering, 2019-10-06:
can you reinstall win10?

Tech-TX, 2019-10-06:
I've never seen that correct a problem that wasn't OS corruption.  In this case, it looks like the IDE is hanging onto the compiled-cached library files for longer than I'd expect, especially when I explicitly asked it not to.

enjoyneering, 2019-10-07:
What IDE do you use?

Tech-TX, 2019-10-12:
The Arduino IDE v1.8.8 with the ESP stuff pulled from the board manager.  I'd tried twice to use the Github pull for the ESP stuff and never could get the ESP libraries to link correctly.

It seems to be working for me now.  I've been doing some extensive testing to see what my problem is, and haven't been able to lock it down.  After swapping libraries with the IDE shut down, then (loading the IDE, closing it and then power cycling the laptop) (repeat) a few times, the error went away.  I hate Win10.  That laptop unhooks the USB serial port every so often when I'm doing multiple compiles & uploads, and I have to power cycle it to get the port(s) back.  My Win7 desktop never does that.

Most things run OK with your latest code!  I've tried around 20 different examples from various people with different I2C modules.  Looks to me like we may be able to mark this 'closed'.  

I have a couple of issues that I'm trying to resolve.   I found a bug in the modified test code above: the #include "wire.h" needed to be #include <wire.h> or I get weird I2C errors during runtime from your I2C library.  The verbose compiler output from both ways of including it looked like it was using your new library both times, but it didn't work correctly with "wire.h" (and that's a consistent error  with multiple re-compiles).  The original wire library doesn't give me those runtime I2C errors.

 I wiped out one of my BME280 sensors during testing by hooking it up wrong - it still works, but the SDA line is pulled low when it's sleeping. :-(

Tech-TX, 2019-10-12:
I finally found out how to clear the old cache junk, and now have a repeatable way to switch libraries.  There were a handful of arduino_build_XXXXXX and arduino_cache_XXXXXX directories under c:\Users\(login name)\AppData\Local\Temp\ that I deleted, as well as a bunch of tmpXXXXXXXX files in that \Temp\ directory after every compile that I'm deleting.  I had > 9500 of them before I started clearing them.

There's a (likely timing) bug in the sample code above that was causing some of my 'data unavailable' errors.  For whatever reason, it causes different problems in the original wire library and your new library.  It turns out I didn't blow up my BME280, it was simply showing a problem in the code.  I've attached a newer version below that shows the problem.  The first Wire.endTransmission() at line 48 is apparently too fast, and is only partially executed.  Without the second Wire.endTransmission() at line 53 the SDA line eventually goes LOW and then alternates every other loop().  The response in the serial monitor is

> Scanning...
> error = 4
> Unknown error at address 0x76

every other loop().  Here's what it does on the logic analyzer:
[image]

With the second  Wire endTransmission() added in, the SDA line goes high again microseconds later and the loop runs OK every pass.
[image]

However, with your library, I see something odd instead of that low pulse:
[image]

Here's that extra write, zoomed in:
[image]

Looks like you have a pointer that's not being cleared after the Wire.write() at line 46.

Here's the full sample code I'm running:

```c

/* --------------------------------------
  Scan and Identify I2C devices by farmerkeith
  21 January 2019  */

#include <Wire.h>

void setup() {
  Serial.begin(115200);
  Serial.print("\nI2C Scan_ID");
  Serial.println("\nScans for I2C devices and names any that it knows about");
  Wire.begin();
}

void loop() {
  delay(1000);           // wait 1 second for next scan
  uint8_t error, address;
  Serial.println();
  //Wire.endTransmission();
  Serial.println("Scanning...");
  address = 0x76;
  //    Serial.print("Status before scan = ");
  //    Serial.println(Wire.status());
  Wire.beginTransmission(address);
  error = Wire.endTransmission();
  Serial.print("error = ");
  Serial.println(error);
  //    Serial.print("Status after scan = ");
  //    Serial.println(Wire.status());
  if (error == 0) {
    Serial.print("I2C device found at address 0x");
    if (address < 16) Serial.print("0");
    Serial.print(address, HEX);
    Serial.print(" !");
  } else if (error == 4) {
    Serial.print("Unknown error at address 0x");
    if (address < 16) Serial.print("0");
    Serial.println(address, HEX);
  }
  if (error == 0) {
    uint8_t data1 = 0;
    Serial.println(" BMP280 or BME280");
    Wire.beginTransmission(address);
    Wire.write(0xD0); // Select ID register
    Serial.print("Error = ");
    Serial.println(Wire.endTransmission());  // Finish I2C Transmission
    Serial.print("Returned number of bytes or 0 (error) = ");
    Serial.println(Wire.requestFrom(address, 1)); // Request 1 byte of data
    Serial.print("Status after read = ");
    Serial.println(Wire.status());
    Wire.endTransmission();  // this causes that LOW pulse (original library) or write (0x3B) yours
    Serial.print("Data available = ");
    Serial.println(Wire.available());
    data1 = Wire.read(); // Read 1 byte of data
    //      Serial.print("Status = ");
    //        Serial.println(Wire.status());
    Serial.print ("Device ID=");
    Serial.print(data1, HEX);
    if (data1 == 0x58) Serial.println(" = BMP280");
    else if (data1 == 0x60) Serial.println(" = BME280");
    else Serial.println(" ID not in list");
  }
}

```

Tech-TX, 2019-10-13:
I've been looking inside the code to figure out why Wire.status() is acting odd.  

Without commenting out the START and STOP,  my logic analyzer's protocol decoder gets confused, as it's never seen that combination before (START immediately followed by a STOP).  Looks like your logic is inverted in twi_status: if SDA is low, try freebus(), and return I2C_SDA_HELD_LOW if it fails.  If SCL is low, try the clock stretch timeout routine, and return I2C_SCL_HELD_LOW if that fails.

Looking at the comments in core_esp8266_si2c.cpp line 618, that's not strictly true.  The freebus() is the way we've done it for years; send 9 clocks followed by a STOP.  It's not a dirty hack, it's recovery if the master got lost or the slave got confused.  Here's a reference to it:
https://www.analog.com/media/en/technical-documentation/application-notes/54305147357414AN686_0.pdf
See Solution 1 on page 2.  I've coded it as a fixed 9 clocks with a STOP, not that "try it and see" they mention in the app note, as you don't know what the slave is doing right then, and if the slave starts sending some 1s then you're right back where you started with a still-confused slave.

Tech-TX, 2019-10-14:
I'll try digging into the **requestFrom(address, bytes)** next to see if I can figure out why Wire.available() isn't returning the number of bytes it read, but I got lost last time I tried.

Just so you know my background, I wrote an I2C library _in assembly_ for a microcontroller around 20-25 years ago 'cos we didn't have one.  I'm good at various assemblers for a few micros, but I'm still a neophyte with C and it's frustrating me beyond belief.

enjoyneering, 2019-10-15:
Thank you for feedback.

You are absolutely right about twi_write_stop() after freeBus(). I will fix it. I also agree with wire.status - it is useless. It was necessary to save it for full compatibility with the original library.

Tech-TX, 2019-10-15:
I'm still struggling trying to debug the library and core file; println() doesn't appear to work in there, so I'm directly manipulating port bits for an "I'm here" method.  It's ugly and slow, but it works.   I miss the emulators / debuggers I used to have for other micros, as I'd have walked it down within an hour with any of them.  Nothing like that for the '8266 that I've seen that would allow debug inside the core & library files.

Tech-TX, 2019-10-16:
This makes NO sense whatsoever.  I added some code inside twi_readFrom() and got no compilation errors.  However, it's not wiggling the GPIOs like I'd have expected it to.  I got a compilation error earlier when I fat-fingered the additions, so I know the compiler sees what I'm doing.  This last pass, zero errors, and suddenly the **error = Wire.requestFrom(address, 3)** in the test program is returning the correct number of read bytes, and **Wire.available()** is also returning the correct number.  However, the GPIOs are stuck low.  They're all initialized as outputs in the test program.  Do I also need to initialize them as outputs inside the core file?  I'd expect the registers to be correctly configured, as they're all low except for one I'm toggling in the test program.  Here's what I added in the core file as test points:
```c
uint8_t twi_readFrom(uint8_t address, uint8_t *buffer, uint8_t length, bool sendStop)
{
  (GPES = (1 << 0));
  delayMicroseconds(10);  //ch3 D2 org GPIO0
  (GPEC = (1 << 0));

  (GPES = (1 << 2));
  delayMicroseconds(10);  //ch4 D3 yel GPIO2
  (GPEC = (1 << 2));

  (GPES = (1 << 16));
  delayMicroseconds(10);  //ch5 D4 grn GPIO16
  (GPEC = (1 << 16));

  (GPES = (1 << 14));
  delayMicroseconds(10);  //ch6 D5 blu GPIO14
  (GPEC = (1 << 14));

  (GPES = (1 << 12));
  delayMicroseconds(10);  //ch7 D6 vio GPIO12
  (GPEC = (1 << 12));
 
  (GPES = (1 << 13));
  delayMicroseconds(10);  //ch8 D7 gry GPIO13
  (GPEC = (1 << 13));
```
That does nothing whatsoever; even the delayMicroseconds(10); isn't doing anything.  The timing is identical with it in or commented out.

Again, I'm *not* a C person yet, so I can't tell if I made an obvious newbie error.  It looks like what you're doing for toggling the pins, but it's easily possible that I'm clueless.  :-)  I still don't know why the Wire.available() is suddenly working, after returning zeroes for a several days.  I got zeros back earlier this evening until the last couple of re-compiles.

If I take the test points out and it still compiles and works, I'll mark it 'closed'.  I wouldn't worry overmuch about the Wire.status() as I can't find anyone using it.  Leave that for a rainy day when you're bored.  I don't have an easy way to test the clock stretch, so maybe I'll order a Nano board and use that as a programmable I2C slave.  I ought to be able to check the clock stretch then with some repeatable numbers.

Update: I commented out the test points above and now the Wire.requestFrom and Wire.available are both returning zeros for number of bytes read again.  I give up.

enjoyneering, 2019-10-16:
Hi Tech-TX,

Added twi_write_stop() after freeBus(). Please check the library .

Tech-TX, 2019-10-17:
Here's what I've been dealing with.  I put the test points into your core file, and also into the sketch, both places just before the Wire.requestFrom / twi_readFrom.  The plot below is what I got for several times of (clear the cache files) (open the IDE) (re-scan the library manager to register any changes) (compile).  No errors in the verbose compiler output, but it flat didn't work (no test points in the core).
[image]

5 or 6 of those cycles later, suddenly the Wire.available is reporting '1' instead of '0', and I get this:
[image]

No difference in the compiler output that I can see, but suddenly it compiled the changes in. GAAAAA!  I feel like Charlie Brown with Lucy pulling the football away from him.

I'll give your new code a shot now, but it may be a while before it actually compiles in.  I just looked at it:
lines 495 & 503 should be doing clockStretch() (check results) instead of **collision - return error**, as stretches almost always happen during the ACK of the address or during the bit read.  I've never seen it during the address bits, although it's theoretically possible for a REALLY slow slave.

and 646-652 (twi_status) should be _something like_
```c
  if (SCL_READ() == LOW) {
     if (clockStretch() != true) return I2C_SCL_HELD_LOW_AFTER_READ              // can't issue a STOP
    }
    /* check if SDA line is blocked */
  switch (freeBus())                  //dirty hack to release locked SDA line
  {
    case I2C_SDA_RELEASED:
      twi_write_stop();               //always use stop after start/unfinished read

      SCL_HIGH();                     //becomes INPUT_PULLUP & pulls line high, SCL FIRST DO NOT TOUCH
      SDA_HIGH();                     //becomes INPUT_PULLUP & pulls line high
      twi_delay(twi_dcount - 18);     //tSU;STA & tBUF >= 4.7μsec is bus free time between STOP & START condition, measured 5.125μsec
      break;

    case I2C_SDA_HELD_LOW:
    case I2C_SCL_HELD_LOW_AFTER_READ:
      return ~I2C_OK;                 //fail to release locked SDA line or stretch time is too short 
      break;
```
so yours would work, even though the original library is non-functional. :-)  Again, *not* a C programmer so my syntax above could be mangled.  I can _usually_ read it, but I'm still learning the ins and outs of C.

enjoyneering, 2019-10-17:
Hi,

> I just looked at it:
lines 494 & 503 should be doing clockStretch() (check results) instead of collision - return error,

It does clockStretch() in twi_read_bit() for 494 & in twi_write_bit() for 503

I have a clone and I use it to check the timings. Found that the 9th bit is always the slowest part - ACK / NACK read.

Tech-TX, 2019-10-18:
Lucky me, it compiled the first time!  I added a line inside the end of freeBus() so I could simulate a stuck bus that got cleared, and see two errors.  One, you're not clocking it 9 times, you exit as soon as SDA appears to be high.  Second, there's a glitch (it looks like somewhere elsewhere in twi_status) that's a total of 1.170us wide from the time it begins, around 990us while it might be read as HIGH by the slave.  I can shoot a scope picture if you want it, but that describes it.  You can see it on the capture below.

My outer program configures the ports as all OUTPUT_OPEN_DRAIN, and I have one pin connected to SDA.  Just before I call Wire.status, I set that pin LOW to simulate a stuck bus, and it's cleared at the end of the first loop in freeBus().  The output is below.
[image]

Here's all I did (one line added to clear the 'stuck bus')

```c
static uint8_t freeBus(void)
{
  if (SDA_READ() == HIGH) return I2C_SDA_OK;                        //SDA is free

  int8_t pollCounter = TWI_I2C_SDA_POLLING_LIMIT;

  /* SDA is low, I2C bus is locked */
  while (SDA_READ() == LOW)
  {
    if (pollCounter-- < 0) return I2C_SDA_HELD_LOW;                 //error handler, ERROR - SDA busy!!!

    SCL_LOW();                                                      //becomes OUTPUT & pulls line LOW
    twi_delay(twi_dcount - 3);                                      //tLOW >= 4.7μsec is LOW period of the SCL

    SCL_HIGH();                                                     //release the SCL bus, so we can read a bit or ack/nack answer from slave
    if (clockStretch() != true) return I2C_SCL_HELD_LOW_AFTER_READ; //stretching the clock slave is buuuuuusy, ERROR - SCL busy!!!
    twi_delay(twi_dcount - 3);                                      //tHIGH >= 4.0μsec is HIGH period of the SCL
    GPOS = (1 << 2);                                                   // end our little 'stuck bus' test :-)
   }

  return I2C_SDA_RELEASED;                                          //SDA released!!!
}
```

The way it's coded, it exits as soon as the first high bit is detected on SDA.  You don't know what state the slave is in, and that might be bit 0 of the data read, so the slave is still lost when the clock stops.  If you won't believe me and Analog Devices on the proper recovery sequence, maybe you'll believe Philips, the inventors of I2C:  https://www.nxp.com/docs/en/application-note/AN10216.pdf  page 17,

> I2C Bus recovery
> • Typical case is when masters fails when doing a read operation in a slave
> • SDA line is then non usable anymore because of the “Slave-Transmitter” mode.
> • Methods to recover the SDA line are:
>   – Reset the slave device (assuming the device has a Reset pin)
>   – Use a bus recovery sequence to leave the “Slave-Transmitter” mode
> • Bus recovery sequence is done as following:
> 1 - Send 9 clock pulses on SCL line
> 2 - Ask the master to keep SDA High until the “Slave-Transmitter” releases
>      the SDA line to perform the ACK operation
> 3 - Keeping SDA High during the ACK means that the “Master-Receiver”
>      does not acknowledge the previous byte receive
> 4 - The “Slave-Transmitter” then goes in an idle state
> 5 - The master then sends a STOP command initializing completely the bus

So, 9 clock cycles with SDA in input_pullup, followed by a STOP.  That's it.  It works.

Don't trust the people that wrote the original Wire library, as they get things wrong as often as the rest of us do. I'll guarantee you that the current implementation won't clear _most_ stuck buses.

I still don't like what you're doing in twi_status.  A status check shouldn't touch the bus if it's OK, it should just look.  You're sending a START - STOP when it should only be a test of SCL and SDA unless either one is found in a low state.

enjoyneering, 2019-10-18:
this is what i found in APPLICATION NOTE AN-686 "Implementing an I2C Reset"

> The method is quite simple. It is the master’s job to recover the bus and restore control to the main program. When the master detects the SDA line stuck in the low state, it merely needs to send some additional clocks and generate a STOP condition. **How many clocks will be needed?** **The number will vary with the number of bits that remain to be sent by the slave. The maximum would be 9.** This number is derived from the worst-case scenario, the case where the processor was reset just after sending an ACK to the slave. Now the slave is ready to send 8 data bits and receive 1 ACK (or NAK in the case of a bus recovery)

Tech-TX, 2019-10-18:
Yes, you're right, my mistake. I just went back and re-checked the spec.  Recognition of STOP is mandatory, so even if the slave is in some weird state it should end the transaction when it sees the STOP condition.  Looking at all of the possible bus error states, that's probably safer than anything else, as the worst case is when the slave halts (communication between master and slave is interrupted) on the ACK of a WRITE address, leaving SDA low.  The (recovery) clocks would be attempting to write 0xFF into whatever register was selected before the interruption.  As long as the slave is responding correctly, it should abort the WRITE when it sees an early STOP condition.

Tech-TX, 2019-10-19:
I have the SDA glitch and extra activity figured out.

When I cause a temporary 'stuck slave', twi_status first calls twi_write_start which calls freeBus(), and freeBus() does a twi_write_stop at line 272 since I've released the 'stuck slave' inside it's loop.  Control returns to twi_status, which immediately issues another twi_write_stop at line 649.  The delay between the two STOPs is 1.170us.  I don't see a way around it the way you have it coded... once you issue the first twi_write_start in a normal (non-stuck) state, you need to follow it with a STOP or you've blocked the bus for any other masters using it.  In the stuck case, you're getting two STOPs.  At a minimum, I'd add a 4 to 5us twi_delay after the STOP in freeBus().

I'm going to try replacing twi_status with the code chunk I suggested here: https://github.com/enjoyneering/ESP8266-I2C-Driver/issues/7#issuecomment-542944839
It'll probably explode with "not in scope" errors until I get that all figured out, which will take several days of googling.  Sigh.  It first checks SCL and attempts to wait it out if stuck, 'cos you can't continue until SCL is clear.  Then it checks SDA, and does a freeBus() if it's stuck.  Return any errors during the procedure, but don't fiddle with the bus if they're both high, just report IT'S ALL GOOD.  From an end-user standpoint, that's what I expect it to do.

I'll continue testing other devices in the background.  The ones I tested earlier (when I had a broken compile) all did blind reads of the buffer without checking Wire.available(), which is why they were working.  I suspect they're all still working.  I do error checking in my code, as I never expect the results to be perfect, especially when dealing with I2C and it's environment.  I'm going to do an extended test with a DS3231 clock chip next, as I've seen that family has lots of issues getting out of sync with the master.  The test code I downloaded for it has *zero* error checking, so it's not surprising the two get out of sync over time. ;-)

Tech-TX, 2019-10-19:
Interesting!  The clock module went into bus lock, and freeBus() hitting it every second with the max number of clocks didn't clear it - ever.  I had to power cycle it to get it working again.  The chip on it is obviously a Chinese clone, as I paid 1/10th the cost of a true Maxim chip for the whole module, which included a CR2032 battery and glue parts.  There's nothing wrong with your code, but the bus didn't recover.  I just researched that problem, and I may need to run the bus recovery at 50KHz for the chip to respond when it's in that berserk state.  The Maxim parts are good to 400KHz bus speed, no telling with a Chinese clone.

I haven't found any other issues, so I'll go ahead and close this one.  **I highly recommend you add a 5us delay after the end of twi_write_stop (see the first paragraph in the previous comment) but other than that your code is working great.**  You also might want to mention the **Wire.setClockStretchLimit(0)** in your README so that people that are looking for a static I2C implementation know that your code supports it.  I haven't tested that, but your changes look OK.

closed 2019-10-19 by Tech-TX

Tech-TX, 2019-10-20:
On sober reflection, I've changed my mind on twi_status.  It should end (after the recovery clocks) with a twi_write_start followed by a twi_write_stop.  That covers the case where there's a stuck slave on the bus with it's SDA pin currently high, which wouldn't be caught otherwise.  While I may not like a 'status' check banging the bus, at least when you say it's all good, it's really all good.  The start-stop should abort anything on the bus if it's still responding.

enjoyneering, 2019-10-21:
Will do. Thank you for help.

dwillmore, 2023-03-31:
I found myself here because of an unrelated google search and stayed to read the whole issue.  Completely worth it. Wow.